TcOpenGroup/TcOpen #29: Scaffold project for Beckhoff's drives.

PTKu, 2021-05-03:


runtimevic, 2021-05-03:
Hello,
 I think it could help to properly develop the component of a motion ptp axis, how can I be part of the team to perform it?
thank you, Víctor.

PTKu, 2021-05-03:
Hola Victor fantastico.  Just sent you an invite to TcOpenGroup org. 
@TomKovac is cooking something, I just spoke to him, when he is able, he will post here the branch where he is working... you can then have a discussion around it. Thanks for the initiative, much appreciated.

runtimevic, 2021-05-03:
> Hola Victor fantastico. Just sent you an invite to TcOpenGroup org.
> @TomKovac is cooking something, I just spoke to him, when he is able, he will post here the branch where he is working... you can then have a discussion around it. Thanks for the initiative, much appreciated.

I am already part of the TcOpen group, if you can add me to TcoBeckhoffDrives and also TcoPneumatics, I think I can collaborate in a positive way, thanks

runtimevic, 2021-05-03:
> Hola Victor fantastico. Just sent you an invite to TcOpenGroup org.
> @TomKovac is cooking something, I just spoke to him, when he is able, he will post here the branch where he is working... you can then have a discussion around it. Thanks for the initiative, much appreciated.
 Hi Peter,
how I can get in touch with TomKovac?
thanks you.

PTKu, 2021-05-03:
@runtimevic I sent you an invite to gitter for a quick comm channel. @TomKovac is there if you need a quick chat you can do it there.

runtimevic, 2021-05-03:
> @runtimevic I sent you an invite to gitter for a quick comm channel. @TomKovac is there if you need a quick chat you can do it there.

I contacted by @TomKovac to gitter
thanks,

closed 2021-05-07 by PTKu