TcOpenGroup/TcOpen #36: example Sequencer logic as you know the missing conditions to move on to the next step?

runtimevic, 2021-05-04:
Hello:
In the example of the Sequencer logic as it is known what are the missing conditions to advance to the next step?
Thanks you, Víctor.

PTKu, 2021-05-04:
Not sure which example exactly, in which folder? Messaging could not be implemented by the component in the example you are looking at.

runtimevic, 2021-05-04:
> Not sure which example exactly, in which folder? Messaging could not be implemented by the component in the example you are looking at.

https://www.youtube.com/watch?v=yCUv2LW-6zo

PTKu, 2021-05-04:
Ahhaa... there are some debug messages, but not full diagnostics implemented in the components, that is something we need to implement.  So for instance, if you go to a pneumatic cylinder you can add timer to time operation time-out and add

```
IF timer.Q THEN Messenger.Error('I did not reach the extremity sensor'); END_IF;
```

In order to detect physical input (sensor, as you may have seen in other inxton videos), we need to conclude decisions in discussion #15.

runtimevic, 2021-05-04:
> Ahhaa... there are some debug messages, but not full diagnostics implemented in the components, that is something we need to implement. So for instance, if you go to a pneumatic cylinder you can add [...]

I think the most successful thing is that the component itself has its own error diagnosis.

RGrabichler, 2021-05-07:
Im very interested in this things and i love this approach to broaden the knowledge about automation, 
because all i do is selftaught, and it is a hell of a mountain to climb.
Everytime i read something like that, i'm getting aware, how much i don't know and cannot even grasp the problems.

Nevertheless i try to contribute:
I've mage my own cylinder fb with diagnosis and i will describe it.
[image]
The In- and Outputs %Q*/%I* are declared in the dut, the linking to hardware in 
[image]
just for reference to [https://github.com/TcOpenGroup/TcOpen/discussions/15](#15).

I have various so called features, which are small fbs. They implement simple task and later can be assembled as components for example as Cylinder or Motor.

My FB Cyl is derived from abstract class Actuator. WIth an Interface IactionProviding i have Methods for fault- handling and the possibility to set pEventid, pSeverity etc.
[image]

Now when instantiating the cylinder, i have the possibility to add timeOuts to WorkPos and/or HomePos.
If the time is set and a timeout occurs, via TCEventHandling System there gets an Event created.

[image]

This is the Diagnose Method
[image]

If this is interesting or useful, i'd appreciate any feedback.

PTKu, 2021-05-08:
Thanks for sharing @RGrabichler! I like your design, it looks very neat. Would you consider the implementation of event logger into TcOpen messaging?

Have a brief look at the following links, we can discuss it in more detail next week.

[conceptual](https://docs.tcopengroup.org/articles/TcOpenFramework/application.html#messenger)

[object](https://docs.tcopengroup.org/api/TcoCore/TcoCore.TcoObject.PlcTcoObject.html#TcoCore_TcoObject_PlcTcoObject_Messenger)

[messenger](https://docs.tcopengroup.org/api/TcoCore/TcoCore.TcoMessenger.PlcTcoMessenger.html), 

Please ignore that it looks like C# documentation we will polish it in the feature, for now just look at member documentation.

[code](https://github.com/TcOpenGroup/TcOpen/tree/initial-dev/src/TcoCore/src/XaeTcoCore/TcoCore/POUs/Prototypes/TcoMessage)

Thanks

RGrabichler, 2021-05-10:
I'd love to contribute to such a project.
As mentioned before, i have hardly any experience, working in a bigger project with other people,
so i'd need a lot of guidance early on.

PTKu, 2021-05-10:
@RGrabichler I'ts OK... I think we are all going to learn along the way.  There is a little tradition in industrial automation to use GH.  We will need to prepare some instructions for the people to get started. I hope we can come with some video this or the next week.