TcOpenGroup/TcOpen #42: We want to have the components usable in applications that do not use full framework.

PTKu, 2021-05-05:
- All components should be usable outside `TcOpen` framework
- Some framework features can be lost (messaging, RPC, etc.)

- [ ] Elaborate pattern for non-framework use
- [ ] Create example(s) of component implementation outside TcOpen framework

PTKu, 2021-05-08:
@Barteling I sketched a way it could work here: 

https://github.com/TcOpenGroup/TcOpen/tree/feature/pkml/src/TcoApplicationExamples/XaeAppExamples/PlcAppExamples/POUs/NonFrameworkUse

The basic idea is that we will need to have something of `ITcoContext` (TcoContextNfCompatibility is in example implementation) declared somewhere (probably global scope) in the application:

~~~iecst
PROGRAM MAIN
VAR
	_context : TcoCore.TcoContextNfCompatibility;	
	_logic : NF_Logic;
END_VAR
//------------------------------------------------------
_context.VolatileOpen(); // OPENS CONTEXT
_logic();                          // ENTRY POINT OF ALL LOGIC
_context.VolatileClose(); // CLOSES CONTEXT
//------------------------------------------------------
+ PROPERTY PUBLIC Context ... Gets _context;
~~~

then NF_Logic block can use the components:
~~~iecst
FUNCTION_BLOCK NF_Logic
VAR
	_piston : TcoPneumatics.Cylinder(NF_MAIN.Context);
END_VAR
//------------------------------------------------------------

_piston.MoveHome();
~~~

Your thoughts?

Barteling, 2021-05-09:
Yep looks, good. As explained I believe this will help for a much higher acceptance of TC open, and I'm not to worried about the empty ITcoContext in the FBInit

Although this painfully displays the lack of support for method overloading in TwinCAT..

PTKu, 2021-05-10:
> Although this painfully displays the lack of support for method overloading in TwinCAT..
👍

closed 2021-05-19 by PTKu