TcOpenGroup/TcOpen #55: Default code style with STweep

Barteling, 2021-05-10:
Last week I discussed the possibility with @PTKu to use STweep to maintain a consistent code style. 
STweep is free for active open source contributors > https://www.stweep.com/opensource/  

So i'd like to start a discussion about using STweep and a default set of settings for TCOpen, 
Do @jozefchmelar  and @TomKovac  use a default code style right now? Or what would be your preference?

RGrabichler, 2021-05-10:
I am using it for a few weeks now, and i like it alot.
Along with agreed Settings of "static code analyser", read on Sagatowskis [https://alltwincat.com/software/](url), this could probably be pretty usefull

jozefchmelar, 2021-05-10:
Hi @Barteling 

I'm using some configuration that my colleague came up with...Since it works, I haven't looked into it😅

Here's the file - [lubo_coding_style.zip](https://github.com/TcOpenGroup/TcOpen/files/6451717/lubo_coding_style.zip)

In most languages I usually use the default settings, whenever possible.
...As far as it looks consistent, it's not a deal breaker for me.

Barteling, 2021-05-10:
Thanks for sharing the settings file @jozefchmelar,
As start the 2 most important settings to discuss are:
'Keep existing line breaks' and 'Existing blank lines to keep'

Which you have at 'true' and '1'.

With these settings you leave some space for a personal touch, 
I personally would prefer to have 'keep existing line breaks' off. 

This results in the most consistent code because STweep will reformat all code as configured

PTKu, 2021-05-10:
I guess we should just take some reasonable formatting and adapt to it. I would prefer not to have a long discussion around this topic. 

@Barteling when it comes to code formatting you're the expert here 😃. So if you share the best formatting settings that you have I personally will just adopt it. What about the others?

Barteling, 2021-05-10:
[Default settings.zip](https://github.com/TcOpenGroup/TcOpen/files/6454886/Default.settings.zip)

I might have learned a bit regarding formatting but it still a matter of taste ^^, any way these are my settings.
Pretty normal I believe. Line length at 150, No alignment in columns, 1 blank line after multi line statements.

And I start the statements in a case on a new line:
```
CASE test OF
    1:
        Statement();
        statement();
    2:
        statement();
END_CASE
```
Let me know your thoughts on this set.

PTKu, 2021-05-11:
see @Barteling PR #57 
[STweep](https://github.com/TcOpenGroup/TcOpen#code-format)

I do agree; do we close here?

closed 2021-05-13 by PTKu