TcOpenGroup/TcOpen #85: Has anyone else had trouble with the runbuild powershell script?

rjgwatson, 2021-05-15:
I have gone through the instructions for setting up the development environment on my host machine and a fresh VM and I get a lot of errors in the CopyInxton stage and the GitVersion state and my build fails.

I have attached the log file from my most recent attempt in a VM. Am I missing a step here?

[PS Build Log.txt](https://github.com/TcOpenGroup/TcOpen/files/6488149/PS.Build.Log.txt)

PTKu, 2021-05-15:
Looks like you are executing the build script from 'C:\Windows\system32\' you should cd to 'C:\Users\richa\Documents\GitHub\'
The errors from 'CopyInxton' are expected in this version of script you can ignore it.
Let me know it that helps.

rjgwatson, 2021-05-15:
Thanks, that gets me around the previous issue. btw you accidentally left a } here:

>   #(GitVersionToPlcVersion $script:gitVersion)
  if( $updateAssemblyInfo){.\_Vortex\builder\uvn.exe -v $plcversion}
}

The next issue that seems to come up is that It looks like the .library files for TcoCore etc are not located in the expected directories (I am unable to find them anywhere):

> C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targe
ts(4919,5): error MSB3030: Could not copy the file "C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCoreConne
ctor\_meta\TcoCore.library" because it was not found. [C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCoreCo
nnector\TcoCoreConnector.csproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targe
ts(4919,5): error MSB3030: Could not copy the file "C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCoreConne
ctor\_meta\version.info" because it was not found. [C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCoreConne
ctor\TcoCoreConnector.csproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targe
ts(4919,5): error MSB3030: Could not copy the file "C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCoreConne
ctor\_meta\TcoCore.library" because it was not found. [C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCoreCo
nnector\TcoCoreConnector.csproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targe
ts(4919,5): error MSB3030: Could not copy the file "C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCoreConne
ctor\_meta\version.info" because it was not found. [C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCoreConne
ctor\TcoCoreConnector.csproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targe
ts(4919,5): error MSB3030: Could not copy the file "C:\Users\richa\Documents\GitHub\TcOpen\src\TcoIoBeckhoff\src\TcoIoB
eckhoffConnector\_meta\TcoIoBeckhoff.library" because it was not found. [C:\Users\richa\Documents\GitHub\TcOpen\src\Tco
IoBeckhoff\src\TcoIoBeckhoffConnector\TcoIoBeckhoffConnector.csproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targe
ts(4919,5): error MSB3030: Could not copy the file "C:\Users\richa\Documents\GitHub\TcOpen\src\TcoIoBeckhoff\src\TcoIoB
eckhoffConnector\_meta\TcoIoBeckhoff.library" because it was not found. [C:\Users\richa\Documents\GitHub\TcOpen\src\Tco
IoBeckhoff\src\TcoIoBeckhoffConnector\TcoIoBeckhoffConnector.csproj]
Error: 15/05/2021 22:16:03:
  .     -v:$msbuildVerbosity    :ErrorsOnly ` `~~~~ [<<==>>] Exception: Exec: Error executing command
build.ps1 failed

PTKu, 2021-05-16:
>Thanks, that gets me around the previous issue. btw you accidentally left a } here:

>>#(GitVersionToPlcVersion $script:gitVersion)
>>if( $updateAssemblyInfo){._Vortex\builder\uvn.exe -v $plcversion}
>>}

Thanks! If fixed that above, sorry for that:)

Could you send me the entire log like before? 

The script should look something like this  under 'Run a multi-line script' https://github.com/TcOpenGroup/TcOpen/runs/2582680982?check_suite_focus=true

It looks like there was some problem creating libraries for some reason. 
Few questions:

- Did the script open Visual Studio (there should be only one instance running)?

- Can you locate file 'C:\Users\richa\Documents\GitHub\TcOpen\_Vortex\builder\vortex.compiler.console.exe'?

- Can you try to run the following?

~~~PowerShell
cd C:\Users\richa\Documents\GitHub\TcOpen
Start-Process .\TcOpen.plc.slnf
.\_Vortex\builder\vortex.compiler.console.exe -s TcOpen.sln
~~~

Thanks for reporting! Let me know please how did that go.

rjgwatson, 2021-05-16:
Strangely, when I ran the script again without doing anything the libraries installed properly. I am not sure what happened last time, but I know that I didn't have Visual Studio open, and one instance opened when I ran the script, so I am not sure what happened before. 

I now get the following error which stops the installation:

> C:\Program Files\dotnet\sdk\5.0.203\Sdks\NuGet.Build.Tasks.Pack\buildCrossTargeting\NuGet.Build.Tasks.Pack.targets(110,5): error MSB4044: The "GetPackOutputItemsTask" task was not given a value for the required parameter "PackageVersion". [C:\Users\richa\Documents\GitHub\TcOpen\src\TcoCore\src\TcoCore.Wpf\TcoCore.Wpf.csproj]
Error: 16/05/2021 10:01:31:
       & $dotnet pack $project -p:PackageVersion=$semVer --output nugets -c $buildConfig /p:SolutionDir=$solutionDir -v     .ildVerbosity --no-restore --no-build
build.ps1 failed

I have attached the full log this time.
[PS Build Log 210516-1003.txt](https://github.com/TcOpenGroup/TcOpen/files/6488935/PS.Build.Log.210516-1003.txt)

PTKu, 2021-05-16:
OK It failed at 'Create packages' that is not necessary for you to continue, that step only prepares the nuget packages for publishing on a feed. The reason why it failed is because we are missing `p:PackageVersion=$semVer` variable that comes from GitVersion step we removed. But you can ignore that.... you really do not need those packages.

Now you should be able to open and work with `sln` (entire solution) or any `slnf` (filtered solutions) files. We will simplify the getting-started script next week #87.  The script in use is doing too many things, it is basically whole pipeline from build to release...

Thanks for reporting again!

rjgwatson, 2021-05-16:
Thanks for the help! It looks like everything is as it should be now.

closed 2021-05-16 by rjgwatson