TcOpenGroup/TcOpen #103: Create GitHub action for an automated build of documentation.

PTKu, 2021-05-18:
- detect change on dev branch
- pull TcOpenGroup/TcoOpen repo
- build documentation in TcOpenGroup/TcOpen.Documentation

PTKu, 2021-05-20:
@jozefchmelar where are we here? Did we test the pipeline?

jozefchmelar, 2021-05-21:
not yet, we had some issues with build, so it was not tested. I'll try to test it today

jozefchmelar, 2021-05-21:
@PTKu  the current blocker is that `dev` branch is write protected on TcOpen.Documentation and it requires a review of 2 maintainers. It looks like it's not possible to create an exception for `TcOpenUser`.

I think that branch protection is not necessary for the documentation repo. 

I could create a new branch with the version name, create a PR using GitHub CLI, and if the review process is skipped it could merge automatically to dev....but this way a lot of complexity would be introduced to otherwise simple script.

PTKu, 2021-05-21:
@jozefchmelar I did remove dev branch protection.

closed 2021-05-25 by PTKu