TcOpenGroup/TcOpen #137: libraries does not find

runtimevic, 2021-05-24:
Hello,
I have installed everything and I keep getting the following error from the libraries, I attach a photo so that it can be seen?
[image]

PTKu, 2021-05-24:
PLC libraries have not been compiled. 
You will need to run the script for the libraries to build and install them into your system. See here: https://github.com/TcOpenGroup/TcOpen#how-to-get-started-with-tcopen. Alternatively, you can open `TcOpen.sln` and run inxton builder with the extension, which should also make the trick... alpha versions of the libraries are released here: https://www.nuget.org/packages?q=TcoOpenGroup but these are not to be used with this solution.

runtimevic, 2021-05-24:
> PLC libraries have not been compiled.
> You will need to run the script for the libraries to build and install them into your system. See here: https://github.com/TcOpenGroup/TcOpen#how-to-get-started-with-tcopen. Alternatively, you can open `TcOpen.sln` and run inxton builder with the extension, which should also make the trick... alpha versions of the libraries are released here: https://www.nuget.org/packages?q=TcoOpenGroup but these are not to be used with this solution.
> Let me know if that helps

@PTKu ,
Now I get that I am missing dll, what do I need to do?

[image]

PTKu, 2021-05-24:
I will need the log from the script if you just could copy-paste it here somewhere. Did it run to the end?

Did you install all pre-requisites from https://github.com/TcOpenGroup/TcOpen#prerequisites? Including NET5?

btw the path at which you have the project is quite long. VS has generally a problem with long paths, there is a max_path 256 https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file?redirectedfrom=MSDN#maxpath
it may not be an issue right now but you could get into trouble in the future.

runtimevic, 2021-05-24:
> I will need the log from the script if you just could copy-paste it here somewhere. Did it run to the end?
> 
> Did you install all pre-requisites from https://github.com/TcOpenGroup/TcOpen#prerequisites? Including NET5?
> 
> I also see that you did not clone the repository, when did you get the copy? We are making constant changes to dev branch so from time to time there could be some problematic commits as we are still in the initial phase. You better clone the repo for faster updates.
> 
> btw the path at which you have the project is quite long. VS has generally a problem with long paths, there is a max_path 256 https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file?redirectedfrom=MSDN#maxpath
> it may not be an issue right now but you could get into trouble in the future.

@PTKu ,
I am using visual studio enterprise 2019 version 16.9.5, is it valid?

PTKu, 2021-05-24:
VS2019 is valid, I do not believe there should a problem...

runtimevic, 2021-05-24:
> VS2019 is valid, I do not believe there should a problem...

@PTKu ,

Gravedad	Código	Descripción	Proyecto	Archivo	Línea	Estado suprimido
Error	CS0234	El tipo o el nombre del espacio de nombres 'PlcTemplateTwinController' no existe en el espacio de nombres 'PlcTemplate' (¿falta alguna referencia de ensamblado?)	PlcTemplateConnector	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\librarytemplate\src\PlcTemplateConnector\Entry.cs	14	Activo
Error	CS0234	El tipo o el nombre del espacio de nombres 'TcoDrivesBeckhoffTwinController' no existe en el espacio de nombres 'TcoDrivesBeckhoff' (¿falta alguna referencia de ensamblado?)	TcoDrivesBeckhoffConnector (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoffConnector\Entry.cs	11	Activo
Error	CS0246	El nombre del tipo o del espacio de nombres 'TcoElementsTwinController' no se encontró (¿falta una directiva using o una referencia de ensamblado?)	TcoElementsConnector (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoElements\src\TcoElementsConnector\Entry.cs	14	Activo
Error	CS0234	El tipo o el nombre del espacio de nombres 'TcoIoBeckhoffTwinController' no existe en el espacio de nombres 'TcoIoBeckhoff' (¿falta alguna referencia de ensamblado?)	TcoIoBeckhoffConnector (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoIoBeckhoff\src\TcoIoBeckhoffConnector\Entry.cs	15	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoPneumatics\src\TcoPneumaticsConnector\bin\Debug\net48\TcoPneumaticsConnector.dll'	PlcAppExamplesConnector	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoApplicationExamples\PlcAppExamplesConnector\CSC	1	Activo
Error	CS0246	El nombre del tipo o del espacio de nombres 'PlcTcProberTests' no se encontró (¿falta una directiva using o una referencia de ensamblado?)	PlcTcProberTestsConnector	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\Tc.Prober\tests\PlcTcProberTestsConnector\Entry.cs	8	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\librarytemplate\src\PlcTemplateConnector\bin\Debug\net48\PlcTemplateConnector.dll'	PlcTemplateTestsConnector	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\librarytemplate\tests\PlcTemplateTestsConnector\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoffConnector\bin\Debug\net48\TcoDrivesBeckhoffConnector.dll'	TcOpen.Group (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\_packaging\TcOpen.Group\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoElements\src\TcoElementsConnector\bin\Debug\net48\TcoElementsConnector.dll'	TcOpen.Group (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\_packaging\TcOpen.Group\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoIoBeckhoff\src\TcoIoBeckhoffConnector\bin\Debug\net48\TcoIoBeckhoffConnector.dll'	TcOpen.Group (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\_packaging\TcOpen.Group\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoPneumatics\src\TcoPneumaticsConnector\bin\Debug\net48\TcoPneumaticsConnector.dll'	TcOpen.Group (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\_packaging\TcOpen.Group\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoffConnector\bin\Debug\net5.0\ref\TcoDrivesBeckhoffConnector.dll'	TcOpen.Group (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\_packaging\TcOpen.Group\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoElements\src\TcoElementsConnector\bin\Debug\net5.0\ref\TcoElementsConnector.dll'	TcOpen.Group (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\_packaging\TcOpen.Group\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoIoBeckhoff\src\TcoIoBeckhoffConnector\bin\Debug\net5.0\ref\TcoIoBeckhoffConnector.dll'	TcOpen.Group (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\_packaging\TcOpen.Group\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoPneumatics\src\TcoPneumaticsConnector\bin\Debug\net5.0\ref\TcoPneumaticsConnector.dll'	TcOpen.Group (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\_packaging\TcOpen.Group\CSC	1	Activo
Error	CS0234	El tipo o el nombre del espacio de nombres 'TcoCoreExamplesTwinController' no existe en el espacio de nombres 'TcoCoreExamples' (¿falta alguna referencia de ensamblado?)	TcoCoreExamplesConnector	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoCore\tests\TcoCoreExamplesConnector\Entry.cs	23	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoffConnector\bin\Debug\net48\TcoDrivesBeckhoffConnector.dll'	TcoDrivesBeckhoff.Wpf (net48)	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoff.Wpf\CSC	1	Activo
Error	CS0006	No se encontró el archivo de metadatos 'C:\Users\Runtimevic\Downloads\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoffConnector\bin\Debug\net5.0\TcoDrivesBeckhoffConnector.dll'	TcoDrivesBeckhoff.Wpf (net48)	C:\Users\Runtimevic\Downlo [... cut ...]
Error	MC3074	la etiqueta 'SampleComponent' no existe en el espacio de nombres XML 'clr-namespace:PlcTemplate;assembly=PlcTemplateConnector'. Línea 13, posición 10.	PlcTemplate.Wpf	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\librarytemplate\src\Wpf\PlcTemplate.Wpf\_Sample\fbSampleComponentManualView.xaml	13	
Error	MC3074	la etiqueta 'SampleComponent' no existe en el espacio de nombres XML 'clr-namespace:PlcTemplate;assembly=PlcTemplateConnector'. Línea 13, posición 10.	PlcTemplate.Wpf	C:\Users\Runtimevic\Downloads\TcOpen-dev\src\librarytemplate\src\Wpf\PlcTemplate.Wpf\_Sample\SampleComponentManualView.xaml	13

runtimevic, 2021-05-24:
@PTKu ,
output is:
Operación Compilar iniciada..
1>------ Operación Compilar iniciada: proyecto: TcoIoBeckhoffConnector, configuración: Debug Any CPU ------
2>------ Operación Compilar iniciada: proyecto: Tc.Prober (Tc.Prober\src\Tc.Prober), configuración: Debug Any CPU ------
3>------ Operación Compilar iniciada: proyecto: PlcTcProberTestsConnector, configuración: Debug Any CPU ------
4>------ Operación Compilar omitida: proyecto: XaeTcoCore, configuración: Debug TwinCAT RT (x86) ------
4>Proyecto no seleccionado para compilarse para esta configuración de solución 
5>------ Operación Compilar iniciada: proyecto: TcoTestRunners, configuración: Debug Any CPU ------
6>------ Operación Compilar iniciada: proyecto: TcoDummyTest, configuración: Debug Any CPU ------
3>E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\Tc.Prober\tests\PlcTcProberTestsConnector\Entry.cs(8,23,8,39): error CS0246: El nombre del tipo o del espacio de nombres 'PlcTcProberTests' no se encontró (¿falta una directiva using o una referencia de ensamblado?)
3>Compilación del proyecto "PlcTcProberTestsConnector.csproj" terminada -- ERROR.
7>------ Operación Compilar iniciada: proyecto: PlcTemplateConnector, configuración: Debug Any CPU ------
8>------ Operación Compilar iniciada: proyecto: TcoPneumaticsConnector, configuración: Debug Any CPU ------
9>------ Operación Compilar iniciada: proyecto: TcoElementsConnector, configuración: Debug Any CPU ------
1>E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoIoBeckhoff\src\TcoIoBeckhoffConnector\Entry.cs(15,37,15,64): error CS0234: El tipo o el nombre del espacio de nombres 'TcoIoBeckhoffTwinController' no existe en el espacio de nombres 'TcoIoBeckhoff' (¿falta alguna referencia de ensamblado?)
1>Compilación del proyecto "TcoIoBeckhoffConnector.csproj" terminada -- ERROR.
1>E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoIoBeckhoff\src\TcoIoBeckhoffConnector\Entry.cs(15,37,15,64): error CS0234: El tipo o el nombre del espacio de nombres 'TcoIoBeckhoffTwinController' no existe en el espacio de nombres 'TcoIoBeckhoff' (¿falta alguna referencia de ensamblado?)
1>Compilación del proyecto "TcoIoBeckhoffConnector.csproj" terminada -- ERROR.
6>Copy C:\Users\Runtimevic\.nuget\packages\inxton.vortex.compiler.console\1.11.1-nightly.487\build\..\tools\vortex.compiler.console.exe to E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\_Vortex\builder
6>Copy C:\Users\Runtimevic\.nuget\packages\inxton.vortex.compiler.console\1.11.1-nightly.487\build\..\tools\GetMyHardwareId.exe to E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\_Vortex
6>Copy C:\Users\Runtimevic\.nuget\packages\inxton.vortex.compiler.console\1.11.1-nightly.487\build\..\tools\uvn.exe to E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\_Vortex\builder
6>TcoDummyTest -> E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoCore\tests\TcoDummyTest\bin\Debug\netstandard2.0\TcoDummyTest.dll
5>TcoTestRunners -> E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoCore\tests\TcoTestRunners\bin\Debug\netstandard2.0\TcoTestRunners.dll
10>------ Operación Compilar iniciada: proyecto: TcoDrivesBeckhoffConnector, configuración: Debug Any CPU ------
11>------ Operación Compilar iniciada: proyecto: TcoCoreTestsConnector, configuración: Debug Any CPU ------
2>Tc.Prober -> E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\Tc.Prober\src\Tc.Prober\bin\Debug\net48\Tc.Prober.dll
12>------ Operación Compilar iniciada: proyecto: TcoIoBeckhoff.Wpf, configuración: Debug Any CPU ------
7>E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\librarytemplate\src\PlcTemplateConnector\Entry.cs(14,35,14,60): error CS0234: El tipo o el nombre del espacio de nombres 'PlcTemplateTwinController' no existe en el espacio de nombres 'PlcTemplate' (¿falta alguna referencia de ensamblado?)
7>Compilación del proyecto "PlcTemplateConnector.csproj" terminada -- ERROR.
13>------ Operación Compilar iniciada: proyecto: TcoCoreExamplesConnector, configuración: Debug Any CPU ------
11>E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoCore\tests\TcoCoreTestsConnector\_generated\StringBuilderTests.g.cs(285,24,285,32): warning CS0114: 'StringBuilderTests.ToString()' oculta el miembro heredado 'object.ToString()'. Para hacer que el miembro actual invalide esa implementación, agregue la palabra clave override. Si no, agregue la palabra clave new.
11>TcoCoreTestsConnector -> E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoCore\tests\TcoCoreTestsConnector\bin\Debug\net48\TcoCoreTestsConnector.dll
11>Compilación del proyecto "TcoCoreTestsConnector.csproj" terminada.
14>------ Operación Compilar iniciada: proyecto: PlcTemplate.Wpf, configuración: Debug Any CPU ------
10>E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoffConnector\Entry.cs(11,41,11,72): error CS0234: El tipo o el nombre del espacio de nombres 'TcoDrivesBeckhoffTwinController' no existe en el espacio de nombres 'TcoDrivesBeckhoff' (¿falta alguna referencia de ensamblado?)
10>Compilación del proyecto "TcoDrivesBeckhoffConnector.csproj" terminada -- ERROR.
10>E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoffConnector\Entry.cs(11,41,11,72): error CS0234: El tipo o el nombre del espacio de nombres 'TcoDrivesBeckhoffTwinController' no existe en el espacio de nombres 'TcoDrivesBeckhoff' (¿falta alguna referencia de ensamblado?)
10>Compilación del proyecto "TcoDrivesBeckhoffConnector.csproj" terminada -- ERROR.
[... cut ...]
40>CSC : error CS0006: No se encontró el archivo de metadatos 'E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoDrivesBeckhoff\src\TcoDrivesBeckhoffConnector\bin\Debug\net5.0\TcoDrivesBeckhoffConnector.dll'
40>CSC : error CS0006: No se encontró el archivo de metadatos 'E:\_BECKHOFF\_BECKHOFF_INXTON\TcOpen-dev\src\TcoDrivesBeckhoff\tests\TcoDrivesBeckhoffTestsConnector\bin\Debug\net5.0\TcoDrivesBeckhoffTestsConnector.dll'
40>Compilación del proyecto "TcoDrivesBeckhoff.Wpf.Sandbox_pefobbg2_wpftmp.csproj" terminada -- ERROR.
========== Compilar: 6 correctos, 30 incorrectos, 2 actualizados, 22 omitidos ==========

PTKu, 2021-05-24:
Si pero eso es de VS2019. Justo? Yo necessito el log del script... tienes que mandarme lo viene fuera de esto:

~~~powershell
cd your_tcopen_folder
.\pipelines\runbuild.ps1 -properties @{"buildConfig" = "Debug";
                                      "isTestingEnabled" = $false;
                                      "msbuildVerbosity" = "minimal";
                                      "publishNugets"= $false;
                                      "updateAssemblyInfo" = $false;
                                      "msbuild" = "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\MSBuild.exe";
                                      "dotnet" = "C:\Program Files\dotnet\dotnet.exe";
                                      "devenv" = "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\devenv.com"}
~~~
es un powershell script!

runtimevic, 2021-05-24:
> Si pero eso es de VS2019. Justo? Yo necessito el log del script... tienes que mandarme lo viene fuera de esto:
>
> ```powershell
> cd your_tcopen_folder
> .\pipelines\runbuild.ps1 -properties [...]

@PTKu , 
Everything is ok, to solve it use the script, doing inxton vortex build does not solve it, the script has been the solution.
thank you, Víctor.

PTKu, 2021-05-24:
The script will place the inxton compiler to the right place it should be in _Vorter\builder\. I just was not probably there....
Will you close this @runtimevic?

runtimevic, 2021-06-05:
Hello,
with the current version it gives errors I put them here:

[image]

PTKu, 2021-06-05:
Moved to #177

closed 2021-06-05 by PTKu