TcOpenGroup/TcOpen #138: app cylinder do not work??

runtimevic, 2021-05-24:
Hello,
The app buttons do not work to send the cylinder to rest or work, and I don't know why?

[image]

PTKu, 2021-05-25:
I believe only the first instance of the cylinder is initialized. That is a sandbox application for testing the UI and so it is not designed to be fully functional. If you want to try you can initialize the other cylinders in `MAIN._defaultContext.Main()` along the lines of the one that is already there.

PTKu, 2022-03-29:
@runtimevic closing here... feel free to reopen if you see it necessary

closed 2022-03-29 by PTKu