TcOpenGroup/TcOpen #153: TcoDiagnosis - diagnosis of software and hardware

philippleidig, 2021-05-28:
Are there already ideas for the implementation of a consistent diagnosis for software and hardware in the PLC? 

I am of the opinion that especially in these areas, a detailed diagnosis is considered to be very useful. 

- IPC / Hardware e.g. CPU usage, ...
- TwinCAT e.g. licensing, realtime info, ...
- EtherCAT / Fieldbus e.g. simple and in-depth
- Motion / Axes

Every EtherCAT slave with CoE can send messages to the TcEventLogger. Only by checking this box. However, these messages are only in German and English. It might be possible to use these messages for diagnostics as well.

[image]

PTKu, 2021-05-28:
This is would be extremely valued @philippleidig.  Is there a way to set the 'enable sending Emergency' programmatically?

philippleidig, 2021-05-30:
@PTKu, the "enable sending emergency" - flag can be activated for example from the PLC via CoE - Write at runtime ( 0x10F3:05 ) 

[image]

I would prefer an EtherCAT diagnosis, for example, without linking variables to the hardware. Or a generic axis diagnosis for CoE and SoE purely based on AXIS_REF. I will give it some thought :)

PTKu, 2022-01-31:
Hi @philippleidig I would leave this open... there have been other priorities but I would like to return to this at some point....

PTKu, 2022-02-22:
@TomKovac this would be nice to consider for the HW diagnostics, would you mind having a look at it?

closed 2022-08-09 by PTKu