TcOpenGroup/TcOpen #178: TcoData MongoDB it doesn't work for me

runtimevic, 2021-06-07:
Hello,
TcoData MongoDB it doesn't work for me.
I put some photos so that it looks like I have it configured:

[image]

[image]

PTKu, 2021-06-07:
It's hard to say... from the pictures. Did you create the `TestDatabase` and `TestCollection` manually?

runtimevic, 2021-06-07:
> It's hard to say... from the pictures. Did you create the `TestDatabase` and `TestCollection` manually?

Hello, @PTKu :
Yes create manually TestDatabase and TestCollection.

PTKu, 2021-06-07:
You modified the unit test code. What is the unit test result when you run it?

runtimevic, 2021-06-08:
> You modified the unit test code. What is the unit test result when you run it?

[image]

runtimevic, 2021-06-08:
> I think we will need to make an (online) meeting with the community to explain how to work with the repo and also we need to go through some concepts about testing (unit, sandbox, and integrations).
> 
> You are running a sandbox application, but the set-up of the repo is made in the unit testing project, you would need to implement the repo in that WPF application for it to work.

yes i think we need it, I'll be waiting for it.
thanks you Víctor

jozefchmelar, 2021-06-15:
Is this issue resolved?

runtimevic, 2021-06-15:
> ¿Se ha resuelto este problema?

Hello, 
I do not know how to add to the wpf the tcodata, hoping to have if there is in youtube or the project some example ...

jozefchmelar, 2021-06-16:
Hola@runtimevic,
good idea about the video, or project example. I'll prepare some example!

jozefchmelar, 2021-06-16:
@runtimevic I tried to write a tutorial here : https://github.com/TcOpenGroup/TcOpen.Hammer/tree/data_examples/HMI/Data 
And the project I'm using : 
https://github.com/TcOpenGroup/TcOpen.Hammer/tree/data_examples 

I'd appreciate your feedback on it... If I'll have some spare time I'll make a video too.

runtimevic, 2021-06-16:
> @runtimevic I tried to write a tutorial here : https://github.com/TcOpenGroup/TcOpen.Hammer/tree/data_examples/HMI/Data
> And the project I'm using : [...]

@jozefchmelar ,
thank you very much, when I can I try it...

closed 2021-08-09 by PTKu