TcOpenGroup/TcOpen #261: TcOpen_the_Vortex_Builder_does_not_exists?

runtimevic, 2021-11-11:
Hello, 
TcOpen_the_Vortex_Builder_does_not_exists?
I attach a photo to show it, how can it be fixed?
[image]

thanks

runtimevic, 2021-11-11:
Hi @PTKu 
I have made a project from scratch to test and anyway I do not believe the libraries in Twincat, I am attaching a photo so you can see it
[image]

I do not know how to fix it...

runtimevic, 2021-11-11:
Hi @PTKu 
I have succeeded, the problem is that the PLC name cannot be given when adding the PLC project,
I attach a photo to show it.

[image]

Now I can continue ...

runtimevic, 2021-11-11:
Hi @PTKu ,
I've gotten stuck again
plc.MAIN._simpleContext, does not contain a definition,
I am attaching a photo so you can see it, how is it fixed?
[image]

PTKu, 2021-11-11:
@runtimevic do you have an instance of SimpleContext in MAIN?
~~~
PROGRAM MAIN
VAR
      _simpleContext : SimpleContext;
END_VAR
~~~

If you do have it then you'll need to run inxton compiler again to refresh the twin project.
Let me know if that helps

runtimevic, 2021-11-12:
> @runtimevic do you have an instance of SimpleContext in MAIN?
> 
> ```
> PROGRAM MAIN
> VAR
>       _simpleContext : SimpleContext;
> END_VAR
> ```
> 
> If you do have it then you'll need to run inxton compiler again to refresh the twin project. Let me know if that helps

Hello @PTKu ,
solved, recompile with Inxton,
Do you know how I can remove the errors from the GitHub project XaeTcOpen101, the first photo I have attached?
thanks

PTKu, 2021-11-12:
@runtimevic  You should replace the string "your-ams-id" with the AMD ID of your target if it is the local you can assign 'null'

so at line 13:

~~~C#
// example:
static string AMS_ID = "192.168.0.2.1.1" ;
~~~

// if your target is local then
~~~C#
static string AMS_ID = null;
~~~

in file `/TcOpen101/TcOpenPlcConnector/Properties/Secrets.cs `

you'll need to fix

~~~ C#
public static class Secrets
{
     public const string pwd = -"your_elastic_password"-;
}
~~~

with

~~~ C#
public static class Secrets
{
     public const string pwd = "runtimevic elastic password";
}
~~~

I left the compile errors intentionally there to force fixing with proper values.

runtimevic, 2021-11-13:
> @runtimevic You should replace the string "your-ams-id" with the AMD ID of your target if it is the local you can assign 'null'
> 
> so at line 13:
> 
> ```cs
> // example:
> static string AMS_ID = "192.168.0.2.1.1" ;
> ```
> 
> // if your target is local then
> 
> ```cs
> static string AMS_ID = null;
> ```
> 
> in file `/TcOpen101/TcOpenPlcConnector/Properties/Secrets.cs `
> 
> you'll need to fix
> 
> ```cs
> public static class Secrets
> {
>      public const string pwd = -"your_elastic_password"-;
> }
> ```
> 
> with
> 
> ```cs
> public static class Secrets
> {
>      public const string pwd = "runtimevic elastic password";
> }
> ```
> 
> I left the compile errors intentionally there to force fixing with proper values.

Hi @PTKu,

If everything you have sent me I already know, the problem is here:
var plc = new TcOpenPlcTwinController (
I am attaching a photo for you to see, that's what I don't know why it's wrong

[image]

PTKu, 2021-11-13:
@runtimevic I thought you were asking about the compile errors from the first item in this thread.

The solution is missing the `*.g.cs` files that are generated by the compiler. If you get the message that the compiler does not exist... it is an issue with the VS. To workaround that problem you just open 'Package manager console' that should copy the compiler into right place. Then you should be able to run it using the extension. When I am able to I will sent you a screenshot.

PTKu, 2021-11-13:
@runtimevic 
Open Package Manager Console:
[image]

In package manage you should find this:
[image]

then run inxton compiler

PTKu, 2021-11-13:
@runtimevic  you can also download the project with all binaries included [here](https://github.com/PTKu/TcOpen-Examples/releases/download/TcOpen101A/TcOpen-Examples.zip)

runtimevic, 2021-11-13:
> @runtimevic Open Package Manager Console: [image]
> 
> In package manage you should find this: [image]
> 
> then run inxton compiler

Hi @PTKu , 
all resolved OK, thank you very much for the quick response, 👍👍😉

PTKu, 2021-11-17:
@runtimevic Yes Victor you can download kibana and elasticsearch from those links... 
Close this issue at your discretion.

closed 2021-11-18 by runtimevic