TcOpenGroup/TcOpen #266: TcoObject and TcoMessenger with TcoCylinder alarms

runtimevic, 2021-11-28:
Hello,

I've been testing the example of: 
https://www.youtube.com/watch?v=amafH75ZiIs (TcOpen 101-2 PLC alarms made easy (really easy). [PLC + .NET C#])
I believe that the diagnosis and the methods of the cylinder component have deficiencies that should be solved.
If the 2 position inputs where the cylinder is located are left active, the 2 cylinder outputs are continuously activated.
for example, as a diagnostic it could be an alarm ("double detection") and no output is activated since there is a double detection and also if there is none of the 2 position inputs, it could be an alarm ("no detection") as a diagnostic. .
Attached photo to show ...
I think the cylinder component needs to be improved so that its diagnosis and methods are better.
.-Another point to take into account is that in the diagnosis itself I do not know if there is a time clock that can be consulted, for example, to know any action how long it takes to execute ...

[image]

Thanks you, Víctor.

PTKu, 2021-11-28:
@runtimevic sure Victor TcoCyclinder component can be improved and expanded. 
The aim of that example is to show the messaging principles. We will come to the components later. I am a bit lost in your description so let me summarize what I understood:

1) **Should it happen that we have both extremity sensors indicating active (logical 1) at the same time we should treat that as a failure of the TcoCyclinder component and we should report a message.** *I do agree.*
2) **None of the extremity sensors indicating in-position should report alarm.** *Here I am not so sure about the utility for practical purpose, but you can change my mind*
3) **Timer to measure the duration of an action.** *Yes. there is timer in TcoTask that provides measurement for of the duration of the task*

PTKu, 2021-11-29:
/cib

runtimevic, 2021-11-29:
> @runtimevic sure Victor TcoCyclinder component can be improved and expanded. The aim of that example is to show the messaging principles. We will come to the components later. I am a bit lost in [...]

Hi @PTKu ,
.-When you never see any of the 2 position inputs in x time, I think you must have the alarm ("there is no cylinder position") and always to perform the action or method of for example moving forward or backward should to have its corresponding detector if not the output cannot be activated, I think it is the most appropriate.
.-In TcoTask, how is time measured? and where it is to see for example a timer of how long each action takes, for example how long it takes for the cylinder to activate the advance output and reach its input that it is advanced.
.- To make a history of alarms, warnings, trace ..., how would it be with TcOpen?

PTKu, 2021-11-29:
@runtimevic 

- From my experience, there are instances when there are no sensors on a piston (often short-stroke cylinders). In principle, you move to one of the two positions and you report the problem when the position is not reached, I believe the states are covered by the 'not reached' message. I do agree that it might be useful information in some cases... is there anyone that would like to share his opinion? What we could do is to raise the message about missing proximity signals optionally.
- `TcoTask` measures the execution time in variable `_elapsedTime` and property `ElapsedTime`
- One way to keep history is to use the context logger, contemplating a solution to persist messages `until acknowledged` if that answers the question.

runtimevic, 2021-11-29:
> @runtimevic
> 
> * Según mi experiencia, hay casos en los que no hay sensores en un pistón (a menudo cilindros de carrera corta). En principio, te mueves a una de las dos posiciones e informas del problema cuando no se alcanza la posición, creo que los estados están cubiertos por el mensaje 'no alcanzado'. Estoy de acuerdo en que podría ser información útil en algunos casos ... ¿hay alguien que quiera compartir su opinión? Lo que podríamos hacer es enviar el mensaje sobre la falta de señales de proximidad opcionalmente.
> * `TcoTask`mide el tiempo de ejecución en variable `_elapsedTime`y propiedad`ElapsedTime`
> * Una forma de mantener el historial es utilizar el registrador de contexto, contemplando una solución para persistir los mensajes `until acknowledged`si eso responde a la pregunta.

Hi @PTKu , 
then the cylinder must be configurable for example with a variable to tell how it is, and accordingly behave according to its configuration. 
you could then have these types of cylinder:
Cyl_1Out_0Pos
Cyl_1Out_1PosHome
Cyl_1Out_1PosWork
Cyl_1Out_2Pos
Cyl_2Out_0Pos
Cyl_2Out_1PosHome
Cyl_2Out_1PosWork
Cyl_2Out_2Pos

PTKu, 2021-12-01:
@runtimevic I added the error message when the cylinder has both sensors active at the same time + some other additions.
as for the error when 'no sensor' detects position after discussion with colleagues we will not implement it at this time. 
this and some additional changes are in #267

PR #267 (2021-12-01): TcoObject and TcoMessenger with TcoCylinder alarms
closes #266

# Pneumatics
- when both extremity sensors of the cylinder are active at the same time error message is now reported the components recovers automatically after solving the detection issue, no restore is required
- added interpolated attributes to allow consumers of TcoCyclinder to rename task names in the UI/HMI
closed 2021-12-01 by PTKu