TcOpenGroup/TcOpen #280: Enable individual Config for Actuator (Cylinder)

RGrabichler, 2021-12-04:
I tried to do and individual TimeOut config for a Cylinder like that.
[image]
in a TcCore.TcoObject, similar to HammerProject
[image]
but it fails with a CheckPointer Fault.
[image]

I used to have a Method, executed on first Cycle, to set Configs, Properties and Names on my Actuators.

Is it possible to do the same here,
or what is the preferred way to use individual Configs.

PTKu, 2021-12-04:
/cib

PTKu, 2021-12-04:
I think in the extended block there is a need to call SUPER^() in the body of the FB
~~~iec
SUPER^(inoAtHomePos := inoAtHomePos, inoAtWorkPos := inoAtWorkPos, inoToHomePos := inoToHomePos, inoToWorkPos := inoToWorkPos);
~~~

@RGrabichler  let me know if that worked for you.

PR #281 (2021-12-06): Enable individual Config for Actuator (Cylinder)
closes #280
closed 2021-12-06 by PTKu