TcOpenGroup/TcOpen #315: Add IdExist, CreateOrUpdate to ITcoCrud

kuh0005, 2022-01-11:


jozefchmelar, 2022-01-21:
/cib

jozefchmelar, 2022-01-21:
already implemented

PTKu, 2022-01-21:
@jozefchmelar  is it or I am missing something?
[image]

jozefchmelar, 2022-01-24:
I'm sorry, it's a bit of a misunderstanding. - No wonder when there was zero info about the issue :) 

The ITcoCrud interface doesn't implement `CreateOrUpdate` or `IdExist`. However, the `TcoDataExchange` does contain implementations for such operations. 

Should `ITcoCrud` even implement such operations?  Right now it lives up to its name. 

We could have an additional interface, which extends from `ITcoCrud` which could implement those operations and even some more in case we need them. Then the TcoDataExchange would implement the extended interface.

PTKu, 2022-06-08:
@kuh0005 is this really needed for some practical reasons?

PTKu, 2022-08-09:
@kuh0005 closing this should it be needed feel free to reopen.

closed 2022-08-09 by PTKu