TcOpenGroup/TcOpen #339: Diagnostics View

RGrabichler, 2022-01-25:
A few things i noticed, which i am used to from other Diagnostics Views

- Auto Updating not working
- Acknowledged (or Rogered as u call it) Timestamp, so open Messsages can be shown
[image]

- in RenderableContentView property, to change default eMessageCategories
- Displaying like the 1000 last Messages with Filtering Function
- Ordering of Messages

Thanks guys

PTKu, 2022-01-25:
/cib

Specter-13, 2022-02-23:
@RGrabichler hey I added some requested features to Diagnostics so if you have time, you can check it out. Look into #356 and #358 .
When you use new sorting table, don't forget to add `services.AddTcoCoreExtensions();` to Startup.cs of your Blazor App.
If you have any questions or ideas, don't hesitate to contact me.

RGrabichler, 2022-02-25:
Thanks, i will look at it soon and give my feedback.

PR #343 (2022-03-04): [blazor]Fixes and feature request for Blazor Diagnostics View
closes #339

from @RGrabichler :

- Auto Updating not working
- Acknowledged (or Rogered as u call it) Timestamp, so open Messages can be shown
image
- in RenderableContentView property, to change default `eMessageCategories`
- Displaying like the 1000 last Messages with Filtering Function
- Ordering of Messages
closed 2022-03-04 by PTKu