TcOpenGroup/TcOpen #390: BlazorApp crashes with "Diagnostics" View

RGrabichler, 2022-03-19:
Blazor crashes with unhandled Exception, when clicking on DiagnosticsView
This is dev-Branch
[image]

This is implementation 
[image]

Specter-13, 2022-03-19:
@RGrabichler @PTKu  Hello, this is not bug, as I wrote in #366, you forget to add to Startup.cs:
 `services.AddTcoCoreExtensions();`
which will enable Fatagrid component in Diagnostics to enable filtering and other features.
Let me know, if this will work.

RGrabichler, 2022-03-21:
@Specter-13  Adding the `services.AddTcoCoreExtensions();` and adding `using TcOpen.Inxton.TcoCore.Blazor.Extensions;` solved the issues.

I noticed, in the TcHammer.Blazor Project this is still missing.
Roman

closed 2022-03-21 by RGrabichler