TcOpenGroup/TcOpen #398: Write fast data as DAQ

patopat, 2022-04-02:
- Implement a buffer for fast data to transmit to the IVC in order to avoid missing data due to the non deterministic task in the C# client. Could be implemented the same way as logger ?

Pat

patopat, 2022-04-03:
/cib

patopat, 2022-04-03:
In case of DAQ the type of data is unknown. So I  will pass a pointer/index thanks to the sizeof on  DAQ struct. The size returned by sizeof() gives the right value. Proof :
[image]

PTKu, 2022-04-04:
Hmm... yeah this is where the generics in the PLC would come in handy... never thought of this approach we may give it a try... we would need then parse the object somehow in .net. (from bytes to concrete objects).

Another possible approach is the one we use `TcoDataExchange` (abstract block) needs concrete implementation, that requires the implementation of _data member that then contains data...

https://docs.tcopengroup.org/articles/TcOpenFramework/TcoData/Introduction.html#getting-started

_data we then work with the _data object on .net side.

A similar approach could be used here where _data would be an array and we could treat it as enumerable when retrieving the data from the buffer.

patopat, 2022-04-04:
I pushed a draft of the TcoDaq in order to open the discussion about the name of files, interface and implementation.