TcOpenGroup/TcOpen #404: Implicit conversion warnings from TcOSequencerBase

rruiter87, 2022-04-22:
There are several implicit conversion warnings as shown below. For example
```
// line 59
_message := CONCAT('ERROR NOT UNIQUE STEP_ID ',UINT_TO_STRING(inStepID));
```
where `inStepID` is actually an `INT`. 

## Solutions

1.  change `UINT_TO_STRING` to `INT_TO_STRING`
2.  change `UINT_TO_STRING` to more general `TO_STRING` 

All warnings:

[image]

| Severity | Code                                                                                                                                | Description                     | Project |
|----------|-------------------------------------------------------------------------------------------------------------------------------------|---------------------------------|---------|
| Warning  | Implicit conversion from signed Type 'INT' to unsigned Type 'UINT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.IsStepIdUnique | 59      |
| Warning  | Implicit conversion from signed Type 'SINT' to unsigned Type 'UINT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)] | TcoSequencerBase.StepInternal   | 34      |
| Warning  | Implicit conversion from unsigned Type 'UINT' to signed Type 'INT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.StepInternal   | 140     |
| Warning  | Implicit conversion from unsigned Type 'UINT' to signed Type 'INT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.StepInternal   | 140     |
| Warning  | Implicit conversion from signed Type 'INT' to unsigned Type 'UINT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.StepInternal   | 150     |
| Warning  | Implicit conversion from signed Type 'INT' to unsigned Type 'UINT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.StepInternal   | 152     |
| Warning  | Implicit conversion from unsigned Type 'UINT' to signed Type 'INT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.StepInternal   | 158     |
| Warning  | Implicit conversion from unsigned Type 'UINT' to signed Type 'INT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.StepInternal   | 158     |
| Warning  | Implicit conversion from signed Type 'INT' to unsigned Type 'UINT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.StepInternal   | 159     |
| Warning  | Implicit conversion from signed Type 'INT' to unsigned Type 'UINT' : possible change of sign [tcocore, 0.5.0.834 (vortex.library)]  | TcoSequencerBase.StepInternal   | 161     |

PTKu, 2022-04-25:
good point @rruiter87, been on my list for some time will have a look

/create_issue_branch

PTKu, 2022-04-25:
/cib

closed 2023-03-21 by peterbarancek