TcOpenGroup/TcOpen #407: TcoInspectors - multiple inspections

kuh0005, 2022-05-23:
Visibility of retry, abort, override tasks according minimal retry value. In case of overAttmepts => e.g. do not show dialog.

When any inspection failed is needed custom mechanic/logical operation in the same calling step. Can be \_currentRunResult member accesible through the property?

Dialog not visible - application start in the moment when sequence is suspended?

Add property for Count "out of range", Time Inside range, Time out of range

[#184] - internal issue

PTKu, 2022-05-23:
/cib

PTKu, 2022-05-25:
> Visibility of retry, abort, override tasks according minimal retry value. In case of overAttmepts => e.g. do not show dialog.

Done with retries normalization in TcoGroupInspector

> 
> When any inspection failed is needed custom mechanic/logical operation in the same calling step. Can be _currentRunResult member accesible through the property?

won't do

> Dialog not visible - application start in the moment when sequence is suspended?

This may happed during commissioning when the app is opened and close frequently
opening separate issue for this here: https://github.com/TcOpenGroup/TcOpen/issues/413

> Add property for Count "out of range", Time Inside range, Time out of range

will do moved to separate issue here https://github.com/TcOpenGroup/TcOpen/issues/414

> 
> [#184] - internal issue

PR #409 (2022-05-25): TcoInspectors - multiple inspections
closes #407

# Group inspection: number of retries normalization
`TcoGroupInspection` now implements the method `NormilizeRetries` that finds the lowest allowed retries on given checks and sets that value to the inspectors used in the group inspection.
Each inspector will run at least once before the normalized retries are updated.
Normalizations must be done before Inspections.

_Usage_
~~~
_groupInspection.WithCoordinator(_coordinator)
		.Act()
		.NormalizeRetries(_diis[0])
                .NormalizeRetries(_diis[1])
                .NormalizeRetries(_dai [... cut]
closed 2022-05-25 by PTKu