TcOpenGroup/TcOpen #414: Inspectors should implement additional information about the progress of the inspection

PTKu, 2022-05-25:
**Is your feature request related to a problem? Please describe.**
Sometimes it is hard to understand what exactly happened during the inspection.
It would be helpful to gather additional information listed bellow. The information should not be exposed to the user, but rather hidden in under additional tab or expander.

**Describe the solution you'd like**
- [ ] counter of out range occurrences
- [ ] overall time spent within the range or required scope
- [ ] overall time spent outside the range or required scope

kuh0005, 2022-05-30:
You can diagnose unstable signal during measurement. Or you can predict problem with hardware.

 **counter of out range occurrences**
  - e.g. higher number can stand for peaks in the analog signal or losing the signal during inspection
 
 **overall time spent within the range or required scope**
 - in case when more inspectors are connected together it can indicate the what the most critical inspection is.
 
 **overall time spent outside the range or required scope**
 - how long you are waiting for signal. Or it can indicate that you are awaiting signal in different range of measurement.

PTKu, 2022-06-04:
/cib

closed 2023-11-29 by peterbarancek