TcOpenGroup/TcOpen #424: [BUG] TcoData - Cannot search for entities saved in MongoDB with special characters saved in _EntityId

JurajSpisiak, 2022-06-06:
**Describe the bug**
When trying to search for entities in DB with prefix such as "(UTC)".

**To Reproduce**
Steps to reproduce the behavior:
1. Save entities in DB with prefix "(UTC)"
2. Try to search for created entities.
4. See error

**Expected behavior**
You should be able to search for entities witch characters such as "+, (, ), /"..

**Desktop (please complete the following information):**
 - OS: Microsoft Windows 11 Pro 21H2
 - Visual Studio version [e.g. Microsoft Visual Studio Professional 2019 Version 16.11.12]
 - TwinCAT3 version [eg. 4024.10]

PTKu, 2022-06-06:
@JurajSpisiak do we use regex there? Would you have a look at it?

JurajSpisiak, 2022-06-28:
/cib

PR #446 (2022-06-29): [BUG] TcoData - Cannot search for entities saved in MongoDB with special characters saved in _EntityId
closes #424
closed 2022-06-29 by PTKu