TcOpenGroup/TcOpen #434: [QUESTION] What the reason for the explicit FB_init call in some function blocks?

rruiter87, 2022-06-15:
I was trying to implement some axis and saw a few warnings about explicit `FB_init` calls. I guess these are not needed?

### Warnings
```
Severity	Code	Description	Project	File	Line	Suppression State
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		TcoMovementState.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		HaltTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		HomeTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		Jog.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		MoveAbsoluteTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		MoveModuloTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		MoveRelativeTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		MoveVelocityTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		Power.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		ResetTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		SetPositionTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		SoEResetTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		StopTask.FB_init	1	
Warning		Method 'FB_init' already called implicitly [tcodrivesbeckhoff, 0.7.0.863 (vortex.library)]		TcoDriveSimple.FB_init	1	
```

PTKu, 2022-06-15:
Right! And excellent question! We have seen that when there was an FB extended by a type in library, we got either compile time error (missing FB_init, in that case the override of FB_init was required) or the FB_init was not called (initialization did not materialize) with this `override` the things did work out. Now this issue started to popup with -2 builds ago and in it looks like now the override is sufficient for not getting compile errors and the FB_init is called implicitly and does what we would expect. 

Creating PR for that 
/cib

PTKu, 2022-06-15:
/cib

PR #435 (2022-06-16): [QUESTION] What the reason for the explicit FB_init call in some function blocks?
closes #434
closed 2022-06-16 by PTKu