TcOpenGroup/TcOpen #468: Logging out by user card

rastislavsimak, 2022-07-21:
Logging out by using an already logged-in card stopped working.(Another card will log it out).

PTKu, 2022-07-21:
The way this is implemented now is that the logout can be done manually (calling logout; pressing the logout method in the app) or by setting logout time, which will log out the user automatically after a preset time.

The idea would then be to log out the user when the logged user reads his own card (token) again?

rastislavsimak, 2022-07-21:
Yes, the idea is to log out the user when the logged user rereads his card. This idea was working after implementation but after restarting the application stopped working.

PTKu, 2022-07-21:
@rastislavsimak when do you need this to be fixed?

rastislavsimak, 2022-07-21:
@PTKu  Im not sure, but its not urgent. We are still debugging the line.

PTKu, 2022-07-21:
@rastislavsimak it is set to https://github.com/TcOpenGroup/TcOpen/milestone/3. Should you need it faster let us know.

rastislavsimak, 2022-07-21:
okay, thank you.

peterbarancek, 2022-08-05:
I would like to request for method **OnAuthenticating**. Something like this  I needed it in my application. There is OnUserAuthenticateSuccess but sometimes you need to check current user and compare with new one and do own scenario. By this  delegate is unable to cover a required customisation... Thanks

PTKu, 2022-08-09:
/cib

PR #486 (2022-08-09): [ENHANCEMENT]Logging out by user card 
closes #468
closed 2022-08-09 by PTKu