TcOpenGroup/TcOpen #519: [NEW-FEATURE] Extend Diagnostics View in Blazor

RGrabichler, 2022-09-26:
This is the Diagnostics View in my Project.
[link](https://freeimage.host/i/sWYKdB)
But i think, there are a few things missing.
I ve got the Timestamp for raised, but i also want Timestamp for Acknowledged.
Icons for the Severity should be displayed for a quick overview.
I prefer the Messages to remain after Acknowledged in the list (perhaps like 1000 items),
sortable like for time raised, time acknowledged and so on.

The Refresh Time  and the Initial displayed Severity should be accessible via Property 
(not sure if already adressed)

Perhaps any other suggestions can be added here

RGrabichler, 2023-09-14:
I tried a few things

- Docker-Compose.yml is created, starts a mongo db to store messages
- 
[image]
- i added property TimeStampAcknowledged to TcoMessage
     - works, but i am unhappy, because it is a breaking change, i will try to revert this
- will add possibilities to config the logging
- Change appearance, filtering etc.

Alternative to the Usage of Messenger, i could change this completly to Display the Logs, so i could display 
every Message from a Module, even if there are pinned Messages Pending.

If there are any suggestions, optimasitions or ideas, im happy to hear

RGrabichler, 2023-09-22:
/cib

RGrabichler, 2023-09-23:
I am still rethinking, if i can use the logger for logging what i want to display as messages.
I have a few issues.

- 
[image]
this is the PlainTcoLogItem.

Here in the TcoLogger.cs
[image]
Can i add the Identity ?

Because i would need this, to check back, if the messenger has active messages.

And i really do not understand fully, what means

PerCycleCount, is this related to category, when from same object more message per cycle came in?
ExpectDequeing, what exactly can i do with this
MessageDigest, some sort of Hashing? does this change when Cycle or TimeStamp changes?

PTKu, 2023-09-26:
@RGrabichler, you could employ 'identity', but it's essential to recognize that each messenger can issue multiple messages. To accurately determine the duration from the logger, you must also consider the content of the message.

Identity essentially represents the pointer value for each TcoObject. This allows us to distinguish the objects within a .NET application.

'PerCycleCount' indicates the number of messages sent by a particular messenger in a single PLC cycle. Though the messenger displays only the most critical message contained in _mime, the logger captures the rising edge of every message occurrence.

'MessageDigest' serves as a checksum for the message content. It alters only when the content of the messages (i.e., text) changes.

RGrabichler, 2023-10-11:
[image]
this is my DB Entry.
The Identity in the 
RenderedMessage : "...  Identity: 18446625899150357168
and the 
Properties.senderlPayload.Identity:  -118174559194448
do not match, looks like a overflow?

PTKu, 2023-10-11:
@RGrabichler is it from MongoDB?

RGrabichler, 2023-10-11:
yes

PTKu, 2023-10-11:
https://www.mongodb.com/docs/manual/reference/bson-types/

Identity is unsigned long (ulong), mongo support only signed long. TcOpen deals with this issue using customized deserializer

PTKu, 2023-10-11:
https://github.com/TcOpenGroup/TcOpen/blob/d1b1696ff7a62b6af1957b9b24fedfc42b68dcac/src/TcoData/src/Repository/MongoDb/Mongo/MongoDbRepositorySettings.cs#L171

closed 2024-05-31 by peterbarancek