TcOpenGroup/TcOpen #524: [QUESTION] Is it possible to relay TcOpen logging to Visual Studio's Error list?

rruiter87, 2022-10-07:
I have some TcOpen objects with loggers. If i want to see the messages, I need to build or open the C# HMI app which is connected to the PLC project. 

Is it also be possible to use the default Error List in Visual studio as a sink for the TcOpen logger? That would be very convenient during development.

PTKu, 2022-10-07:
This is the list of available serilog sinks that I am aware of.

https://github.com/serilog/serilog/wiki/Provided-Sinks

Sinking the messages to VS would probably require using DTE or VS extensibility. 
Possible, but there is nothing out of the box.

What you could do, though, is to have a simple console application without full HMI.

Something along the line:

~~~CSharp

class Program
    {
        static void Main(string[] args)
        {
            PlcConnector.Entry.Plc.Connector.BuildAndStart().ReadWriteCycleDelay = 10;

            TcOpen.Inxton.TcoAppDomain.Current.Builder
                           .SetUpLogger(new TcOpen.Inxton.Logging.SerilogAdapter(new LoggerConfiguration()
                                                   .WriteTo.Console() // Writes to console
                                                   .WriteTo.Notepad() // Writes to openen instance of notepad.exe
                                                   .WriteTo.File(new Serilog.Formatting.Compact.RenderedCompactJsonFormatter(), "logs\\logs.log") // Write to file
                                                   .MinimumLevel.Verbose())); ;

            PlcConnector.Entry.Plc.MAIN._myContext._logger.StartLoggingMessages(TcoCore.eMessageCategory.Info);

            Console.Read();
        }
    }
~~~

required packages
~~~XML
<ItemGroup>    
    <PackageReference Include="Serilog.Sinks.Notepad" Version="2.1.0" />
	<PackageReference Include="TcOpen.Group" Version="0.10.0-alpha.5" />
  </ItemGroup>
~~~

Not sure if this will do...

You can let the application run, it should recover from transition stop/run/config states automatically.

rruiter87, 2022-10-26:
Thanks for the answer! Closing.

closed 2022-10-26 by rruiter87