TcOpenGroup/TcOpen #531: [BUG] TcoInspector Failure Description and Failure Codes are not working as expected

KrisztianOlah, 2022-10-18:
**Describe the bug**
Calling Inspector.Data.FailureDescription or Inspector.Data.FailureCodes properties will return an empty string when descriptions and codes are saved in Process Data

**To Reproduce**
Steps to reproduce the behavior:
1.Save Failure descriptions and codes to recipe
2. Load recipe to plc 
3. Call Inspector.Data.FailureDescription and Inspector.Data.FailureCodes
4. See error

**Expected behavior**
They both should return a concated list of descriptions and codes of all errors occurred

**Screenshots**
[image]

PTKu, 2022-10-18:
Thanks @KrisztianOlah 

I am unable to reproduce.

When you get the chance, can you check if the `ProcessData` structure (the one connected to the recipe) gets the data correctly?

I checked that `plain` -> `shadow` -> `online` are transferred correctly. 
The concatenation of the failure messages also seems to work.

KrisztianOlah, 2022-10-18:
Hi Peter, sorry my bad, I didn't remember my use case well. The above example works, however, I was wondering if it was possible to access the list of failures via THIS^.ProcessData.EntityHeader.Results.Failures (Cu), because this did not work for me. Perhaps I'm missing something. I would like to access a list of all the failures.

PTKu, 2022-10-18:
If you mean the list of failures (string, semicolon separated failure codes that occurred on given entity) they are in the entity header in your case `ProcessDataManager.Data.EntityHeader.Results` failures. Is this what you need?

KrisztianOlah, 2022-10-18:
Yes, you are indeed correct. However, it still returns an empty string for me even though accessing the inspector directly returns the correct string.

PTKu, 2022-10-18:
Will need to see the implementation in your application. Will reach out over internal channel then we post the solution here.

closed 2023-03-21 by peterbarancek