TcOpenGroup/TcOpen #546: [BUG] VS crash TcOpen

nico10hahn, 2022-10-31:
**Describe the bug**
Visual Studio crashes when compiling the TcOpen project from the introductory video ("TcOpen 101-1 Introduction to TcOpen framework TcoContext and TcoLogger (PLC + .NET C#)"). TcOpen.Group is installed via NuGet and the Inxton compiler is installed and licensed.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow all instruction of the Video
2. Click Inxton-Vortex-Build

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 11 Pro 22H2 ]
 - Visual Studio: Microsoft Visual Studio Community 2019 version [Version 16.11.20]
 - TwinCAT3 version [v3.1.4024.35]
 - Microsoft .NET Framework Version 4.8.09032

PTKu, 2022-10-31:
Hi @nico10hahn, Thanks for reporting!
I am unable to reproduce. It could be there is a problem with the language mutation of VS (de), the project was created in `en-US`. It gave us problems in the past (I suspect there is a problem searching the project tree `TcOpenPlc Projekt` vs `TcOpenPlc Project`). Would you be able to run VS in English, and let me know if the project persists?

nico10hahn, 2022-10-31:
Hello @PTKu, thanks for the reply. You were right, it works fine when VS is set to english. Many thanks for the help !

closed 2022-10-31 by nico10hahn