TcOpenGroup/TcOpen #548: [QUESTION]

nico10hahn, 2022-11-07:
Send messages from Beckhoff to console log:

I used the example with the PlcConnector to write simple console logs (it works). Is it possible to adapt the example and have console logs written by another controller instead of a local application ? How would the environment variable (Tc3Target) have to be set for this ?

Thanks for help

PTKu, 2022-11-07:
Hi @nico10hahn 

you can replace `Tc3Target` with a string of the AMS ID (e.g. "192.168.0.1.1.1") of your target system. Or change/add the `Tc3Target` environment variable to your environment. Remember that you will need to restart visual studio for the environment variable to be loaded.
Let me know if it works for you.

nico10hahn, 2022-11-10:
Hello @PTKu ,

thanks. Now its working ! Very cool .

closed 2022-11-10 by nico10hahn

nico10hahn, 2022-11-10:
Thanks for support and helping,
its also possible to use TcOpen to write files (like Serilog.Sinks.Notepad) or console logs from the Beckhoff soft sps to the user space of the system where windows is running (or only on the device where VS Studio is running)?

PTKu, 2022-11-10:
The logs are created by the application. So no, you don't need the visual studio for it to work.