TcOpenGroup/TcOpen #577: [BUG] TcoTask.Busy is active prematurely, invoke needs to be called multiple times

mhareznik, 2023-01-30:


mhareznik, 2023-01-30:
For more details:
Declaration of _parallelTask:
`_paralellTask : CU00xAutomatParalell_1_Task(THIS^, TcoCore.eRestoreMode.AutoRestoreMembers);`

Cyclic calling of _parallelTask in CU00xAutomatTask.Main method:
`_paralellTask();`

Invoking task in step of sequence:
`IF (Step(2004, TRUE, 'START PARALELL TASK')) THEN`
`//-------------------------------------`
`	StepCompleteWhen(_paralellTask.Task.Invoke().Busy);`
`//-------------------------------------`
`END_IF`

PTKu, 2023-01-30:
You will need to use 'TcoCore.eRestoreMode.None' instead of 'TcoCore.eRestoreMode.AutoRestoreMembers', when you want to use the task asynchronously.

[image]