TcOpenGroup/TcOpen #601: [QUESTION]

soup01Threes, 2023-04-05:
Dear,

I would like to try the TcOpen Framework,
But after I install the TcOpen.Group and starting to build in my project,
this error is shown:

`The attribute "xmlns" in element <ItemGroup> is unrecognized.	plc1Connector	C:\Users\root\Source\Repos\TwinCAT Project1\plc1Connector\plc1Connector.csproj	396	`

Would you give me some hints about this?

Thanks and BR,

PTKu, 2023-04-05:
Hi, thanks for reaching out. Can you share file 'plc1Connector.csproj' with us?

soup01Threes, 2023-04-05:
> Hi, thanks for reaching out. Can you share file 'plc1Connector.csproj' with us?

Thanks ！
And here is the file:
[plc1Connector.csproj.txt](https://github.com/TcOpenGroup/TcOpen/files/11165007/plc1Connector.csproj.txt)

PTKu, 2023-04-06:
> > Hi, thanks for reaching out. Can you share file 'plc1Connector.csproj' with us?
> 
> Thanks ！ And here is the file: [plc1Connector.csproj.txt](https://github.com/TcOpenGroup/TcOpen/files/11165007/plc1Connector.csproj.txt)

You will need to use newer `csproj` file fromat (https://learn.microsoft.com/en-us/dotnet/core/project-sdk/overview)

The format used in your project is not supported by the inxton compiler.

It also looks like you're adding the PLC twins to a ASP.NET project. It would be better to create a 'library' project that you'll reference in the projects you need.

To create connector project use `.NET standard / .NET core` format instead of `.NET Framework`
[image]

soup01Threes, 2023-04-06:
Hi Mr PTKu,

Thanks for your support and I can imported the TcOpen Library into my TwinCAT Project without Error.
Thanks for your Support and I would like to close this issue;)
[image]

soup01Threes, 2023-04-06:
The problem is solved.

closed 2023-04-06 by soup01Threes