TcOpenGroup/TcOpen #610: [NEW-FEATURE]Analog inspector Master/Dummy mode

KrisztianOlah, 2023-05-11:
**Premise**
When setting up process settings for master references one has to set all inspectors to exclude result in order to check which way it needs to fail. Then set every inspector with the correct "failed" range. On one project I have 15 analouge ranges per reference with 4 references with some designed to fail whilest some destined to pass due to physical/mechanical constraints which took about an hour to set up for all references.

**Proposed solution**
A tick box on analouge inspector's UI to flip the result for Master pieces on demand, which will use the same range as the OK parts but will fail if it is within range and will pass if it is outside of range which should markedly reduce (programmer) time spent on production on tester stations. On balance this would be a simple a quick solution that would also be very useful for the user.

**Alternatives considered**
- using continue and evaluating fail results individually. On my project this solution would require operator to keep a table describing what dummy should fail what tests, which is unfeasable. Also, it would require additional code as this goes against the requirement of not continuing after a failed test.
- adding relational signs (in a combobox for simplicity) to process data to more accurately describe what ranges are acceptable. This would complicate UI, but would be workable. It would require considerable work on TcoInspectors though.
- Changing behaviour of failing a test to show failed test, but also marking it correct in case of some dummy references. This could be a larger overhaul, but usage would more intuitive.   

I'm happy to work on this provided we can determine the ideal direction after some constructive discussion.

PTKu, 2023-05-11:
This popped up several times already... we should do it... @peterbarancek?

KrisztianOlah, 2023-05-12:
/cib