TcOpenGroup/TcOpen #639: [Suggestions] Suggest UTC Now as standard Time Format, confused by logging times atm

RGrabichler, 2023-09-09:
In the Issues #519 ]New-Feature] Extended Diagnostics in Blazor, i tinker around with the TcoMessages, i i noticed a TimeStamp Issue.

[image]
this is Timestamp logged in Console

[image]
this is the Timestamp in the Plc

[image]
this is my local Timestamp in Germany and Summertime

[image]
this is UTC

And this is a complete mess in my opinion.

I would suggest, to allways log in UTC, and everybody can display whatever he wants.

I only assume, the 15.50 timestamp comes from local Slovenia Time?  UTC+2 ? In difference to Berlin +1 ?

So in my opinion, the Logger should use per default UtcNow, the Connector should use UtcNow, and 
the Wpf or Blazor View can render whatever they want for the User, probably Utc with corresponding TimeZone.

PTKu, 2023-09-09:
There should not be a time difference between the time zones you mention.

@TomKovac?

@RGrabichler could you check that you call time synch in your program for the context? Something like this:
https://github.com/TcOpenGroup/tcopen-app-templates/blob/ec2b13f96b091538887fa9851f5fe3d3adfcdeec/templates/mts-s-template/t/src/x_template_x-xae/x_template_xPlc/Technology/Technology.TcPOU#L109C1-L109C29

RGrabichler, 2023-09-09:
[image]
i call this here
[image]

PTKu, 2023-09-09:
You're right. It should be aligned... will have a look on Monday.

PTKu, 2023-09-09:
/cib

PTKu, 2023-09-12:
@RGrabichler, what are the region settings of your PLC (time-zone)?

RGrabichler, 2023-09-12:
for me it is UTC+1, Berlin Time.

PTKu, 2023-09-12:
Here's the fixed text:

"Same as me! 😊
Where do you run your Blazor server? Is it on the PLC, or is it hosted elsewhere? It appears that there might be a difference in the environment settings between where you run your server and the PLC.

You can try running it with the second argument, specifying the AMS ID of your server as follows: `app.RtcSynchronize(TRUE, 'YOUR_SERVER_AMS_ID', 60)` (An empty string indicates that it will synchronize with the local system)."

PTKu, 2023-09-12:
"Same as me! 😊
Where do you run your Blazor server? Is it on the PLC, or is it hosted elsewhere? It appears that there might be a difference in the environment settings between where you run your server and the PLC.

You can try running it with the second argument, specifying the AMS ID of your server as follows: `app.RtcSynchronize(TRUE, 'YOUR_SERVER_AMS_ID', 60)` (An empty string indicates that it will synchronize with the local system)."

RGrabichler, 2023-09-12:
ok, in this specific scenario, where i tinker with the Diagnosis Component, i use UmRT and blazor runs on local Dev Pc/
So presumably both run on UTC +1.

PTKu, 2023-09-14:
It looks like that might be an issue with UmRT. 
Did you try:
```
You can try running it with the second argument, specifying the AMS ID of your server as follows: app.RtcSynchronize(TRUE, 'YOUR_SERVER_AMS_ID', 60) (An empty string indicates that it will synchronize with the local system).
```
?

RGrabichler, 2023-10-02:
i tried, and i tried it with real hardware.
both have identical time settings
[image]
[image]
[image]
this is AMS ID from dev PC, were i run the Blazor App atm

PTKu, 2023-10-03:
@TomKovac any comment on this?

peterbarancek, 2023-10-03:
"This is my output. I am running an app on my laptop. The time is synchronized with the PLC (as seen in the screenshot at the bottom right of the picture). It runs on .net48 and is a WPF application. In my opinion, the time difference might be due to using .net5 or Blazor..."

[image]

[image]

RGrabichler, 2023-10-04:
> @TomKovac any comment on this?
> 
> @RGrabichler sorry if you answered this question before, but what would be your correct local time in the example above?
> 
> @peterbarancek what time do we set to the TimeStamp in the messenger?

in this case, 21.44 was the right time.

RGrabichler, 2023-10-04:
i tried a bit 
[image]
[image]
this is the ams ID from the PLC, manually changed the Clock to differ from real Time, to test it out.
[image]

This is my Dev PC 
[image]
but this is not working, what am i doing wrong?
Time shows still Plc Time, even when i am setting AMS ID to Dev PC, where Blazor runs.

PTKu, 2023-10-05:
@RGrabichler update rate is 60 seconds, you can adjust the second parameter to lower the rate.

Back to the original problem. We found the reason for the time difference. Connector.NET5 uses a newer driver of ADS, which takes the time zone into account. Since the time in the PLC is flagged UTC, when the value is read from the controller, it is adjusted to the local time.

Unfortunately, we do not have an immediate solution to this situation. It will require changes in the inxton connector (which is experimental and not officially supported for .NET 5 target). 
We can make adjustments when releasing major updates of inxton that will take place past TwinCAT 4026 release.

RGrabichler, 2023-10-05:
ok, im fine with that and can implement a workaround with manually adding or subracting hours :X
thanks for the effort

closed 2023-10-05 by RGrabichler