TcOpenGroup/TcOpen #642: [NEW-FEATURE] Port to dotnet6/7 with new TwinCat 4026 for Blazor

RGrabichler, 2023-09-13:
I would love to be able to use the new Dotnet Versions with Visual Studio 2022 and the TC4026.
I really have no clue, how much work and changes are behind this.

peterbarancek, 2023-09-14:
Our commitment is to keep supporting .NET Framework 4.8. And we will make it work with 4026, not sure yet how many nasty surprises we get.
About dotnet6/7/8. TcOpen is simple; The communication layer of Inxton technology must be refactored and tested in detail. We don't have resources for this at this point as most of the team moved to AX# and AXOpen.
Should Beckhoff start providing proper release notes and begin seeing contributors to the TwinCAT ecosystem as partners by sharing information as we are used to in the software world, we may start considering allocating some resources for that 😉.

peterbarancek, 2024-05-30:
We only tried the beta version to verify if our pipeline is working. However, there is a plan to move to 4026, maybe after summer.