TcOpenGroup/TcOpen #713: [BUG] Correct User Level Assignment

mhareznik, 2024-05-14:
In order to ensure the integrity and security of our system, it's essential to assess the security of the library TcOpen.Inxton.Local.Security.Wpf used for user management. User level is not correctly assigned. Looks like, problem is after the addition of a new user or generating a default one.

[image]

Also, there is an inconsistency in the displayed created and modified time information within our system. The displayed timestamps do not accurately reflect the actual creation and modification times of various entities, which can lead to confusion and incorrect assumptions about the data's freshness.
[image]

peterbarancek, 2024-05-23:
/cib

peterbarancek, 2024-05-23:
/cib

peterbarancek, 2024-05-23:
/cib

closed 2024-05-29 by peterbarancek