TcOpenGroup/TcOpen #718: [NEW-FEATURE] Add MC_TorqueControl

rruiter87, 2024-06-05:
Is it possible to add [`MC_TorqueControl`](https://infosys.beckhoff.com/content/1031/tcplclib_tc2_mc2/7617393803.html?id=228837685507697346) to TcOpen? I've started to add it to a project that I'm currently working on, but I find working with TcOpen quite overwhelming. 

I can open a PR with what I have so far, maybe you can provide some pointers on what is missing or things I did wrong.

peterbarancek, 2024-06-05:
It is up to you. If you want, you can prepare your content and share with me. I will review it. Or wait  i will prepare it for you.

rruiter87, 2024-06-07:
See #720 

I tried setting up the required development environment, but I can't download the visual studio 2019 community version installer anywhere. With Visual Studio 2022, the Inxton compiler plugin refused to install.

peterbarancek, 2024-06-12:
/cib

rruiter87, 2024-06-14:
Thanks for implementing this so fast!

PR #722 (2024-06-12): [NEW-FEATURE] Add MC_TorqueControl
closes #718