TcOpenGroup/TcOpen #726: [QUESTION] Is it possible to use TcUnit with TcOpen objects?

rruiter87, 2024-06-18:
Is there a way to use TcUnit with TcOpen objects? I now tried the following. I have this sensor I want to unit test, that derives from TcoComponent.

```
FUNCTION_BLOCK SomeSensor EXTENDS TcoCore.TcoComponent
...
```

When I instantiate it for the unit tests, I need to pass a reference to the parent object to it. But that doesn't compile.

```
FUNCTION_BLOCK LinearQuadraticRegulator_Tests EXTENDS FB_TestSuite
VAR
    /// either instatiate it here
    sensor : SomeSensor(THIS^);
END_VAR

METHOD TestInputValidation()
VAR_INST
    /// or here
    sensor : SomeSensor(THIS^);
END_VAR
```

I also tried to instantiate a TcoComponent in the sensor function block. But also this doesn't compile.

```
```
FUNCTION_BLOCK SomeSensor 
VAR 
    _core : TcoCore.TcoComponent(THIS^);
END_VAR
...
```

Is there a way to do this?

RGrabichler, 2024-06-18:
``Yes it is, PTKu showed me a way.
You have to Impelemnt TcoCore.ITcoContext. like that
```
FUNCTION_BLOCK LinearQuadraticRegulator_Tests EXTENDS TcUnit.FB_TestSuite IMPLEMENTS TcoCore.ITcoContext
VAR
  _nfContext : TcoCore.TcoContextNfCompatibility; //Context for TcOpen
END_VAR
```

Then Add a Implementation of this to the tested Function Block.
[image]

Add the Implementation like that
`Context := This^._nfContext;` etc..
Then u can proceed like always with the tests

PTKu, 2024-06-18:
https://github.com/tcunit/TcUnit/discussions/199#discussioncomment-4315931

rruiter87, 2024-06-19:
Thanks for the elaborate answer @PTKu!

closed 2024-06-19 by rruiter87