TcOpenGroup/TcOpen #731: [IDEA] Fix mix of TcVersions

rruiter87, 2024-06-19:
When I open the TcOpen main solution file, I get many warnings popping up about different TwinCAT versions that are not installed on my system. 

[image]

Here is a screenshot of several versions. I'm not sure what the best course of action is to prevent all these pop-ups. Maybe through a pre-commit hook?

[image]

PTKu, 2024-06-19:
Yeah, this is a major annoyance. @rruiter87, would you be so kind as to have a look at the pre-commit look? If it's somewhat easy to take, you can do it. Please get in touch with @peterbarancek if you need some support. I am not sure, but I think the build script does some modifications to tsproj files already to align with versions... that might also be the way.

rruiter87, 2024-06-19:
Yeah, I can take a look at the pre-commit hooks. If we decide to go that way, I want to add it to the [existing TwinCAT pre-commits](https://github.com/pcdshub/pre-commit-hooks), if the maintainers agree.

In your build scripts, it only removes a pinned version if present:
https://github.com/TcOpenGroup/TcOpen/blob/803996ad3a01f75324bd3491fd3b5f00b9b1f6fa/pipelines/build.ps1#L105

Since this is in the build script, a pinned version can still be present in the committed code. 

A pre-commit would anyway be nice, then I can also include the STweep formatting for example.