TcOpenGroup/TcOpen #741: [QUESTION] Blazor Representation of TcoDriveSimple not working

RGrabichler, 2024-08-22:
I try to use the TcoDriveSimple in my Plc with NC and I have Problems with the Representation in Blazor.
[image]

The Plc Side is working.
Now when i try to use a minimalistic Example
[image]
i get this, but cannot expand it.
[image]

Not sure, what can be missing.
In PlcHammer.Hmi.Blazor it is working, i cannot find the critical difference.

PTKu, 2024-08-22:
Hey @RGrabichler, I think you will need to reference
https://www.nuget.org/packages/TcOpen.Group.Blazor/
or
https://www.nuget.org/packages/TcOpen.Inxton.TcoDrivesBeckhoff.Blazor/

in your hmi project.

RGrabichler, 2024-08-23:
[image]
the TcOpen.Group.Blazor was already referenced, and this should include the TcoDrivesBeckhoff.
But nevertheless, i also referenced the TcoDrivesBeckhoff.Blazor, but no success.

PTKu, 2024-08-23:
Hmm... check if you added following service

```
services.AddVortexBlazorServices();
```

PTKu, 2024-08-27:
Weird... can you share the project?